OpenX6100/x6100_gui #3: Wi-Fi not active in firmware. Feature request.

jasonmtravis, 2024-09-25:
Was on Oleg’s roadmap for the build. Please continue.

Thanks, 73
W5jmt

gdyuldin, 2024-09-26:
Right now adding WiFi support will only make possible synchronization transceiver time through NTP.
I have no other things in the backlog, which requires network connection - so this task has low priority.
Perhaps, you can suggest some new features, that's requires WiFi?

73. Georgy // R2RFE

jasonmtravis, 2024-09-26:
One of the most fascinating things about an open source network connected radio are the options. 
Probably the most realistic would be OTA updates.
QRZ.com callsign lookups.
SSH access
AX.25 routing for remote packet networks (basically making the X6100 into a wifi hotspot for packet) to stand up a network when there's no cell service. emergency stuff.

These are just a stream of consciousness, and I have no idea how feasible they would be considering the resources that the X6100 has. 

73,
W5JMT

W8LID, 2024-09-28:
I would think NTP sync would be justification enough for WiFi. Time sync is critical for FT8 and the current UI to set time is cumbersome and often inaccurate. Being able to SSH into the radio was neat but did not find any practical applications for it yet. Another potential use would be to automatically post to PSKReporter?

Thanks for the work on this firmware, it seems pretty nice overall so far.

73,
W8LID

gdyuldin, 2024-10-22:
Added in 0.26.0

W8LID, 2024-10-22:
Thanks for implementing this. NTP sync works great, it’s nice to not have to fix the system clock every time I power up my radio.

jasonmtravis, 2024-10-24:
Cheers Georgy!

73, 
Jason - W5JMT

gdyuldin, 2024-11-27:
I'll try to use code from wsjtx to make a better FT8/FT4 decoding and, perhaps, WSPR decoding too.