OpenX6100/x6100_gui #4: switching mode via CAT leaves old filter (CW -> USB-D)

jflinchbaugh, 2024-09-25:
If I'm running CW mode with its default narrow filter, and then I have wsjt-x tell flrig to switch frequencies to go do some FT8, it switches frequency and it sets the mode to USB-D via CAT, like it should, but the filter stays narrow like it's still in CW mode. The radio display says USB-D. I can get it back to the normal filter width with the SSB button on top, toggling to LSB-D and back to USB-D. I'm running 0.24.1 now, and this misbehavior seems to have cropped up maybe in 0.24.0. I'm not positive, but it used to automatically switch filters with mode.
Thanks.

jflinchbaugh, 2024-09-25:
more detail: the radio UI says the filter has been restored, but the digital audio output to the computer and the audio to the speaker are still constrained to 550-850 Hz.
Switching frequencies a second time in wsjt-x, either FT8/FT4 or by band, in wsjt-x also fixes the filter on the second switch.

jflinchbaugh, 2024-09-25:
also, i'm really happy we can report bugs out here on github instead of trying to follow confusing threads on the groups.io mailing list. :+1:  It's great to have a nice place here to download the full builds for each release. I really appreciate the work! Thanks.

gdyuldin, 2024-09-26:
Thanks for your report. I'll take a look. Perhaps, something is just missed on CAT implementation on this firmware.

jflinchbaugh, 2024-09-26:
It also reproduces if the radio was previously in a "phone" mode (USB/LSB) and CAT switches to a "digital" mode (USB-D/LSB-D). It leaves the audio bandwidth at what it was for phone mode, usually 2700hz or 2800hz, instead of expanding to 3000hz.

closed 2024-09-30 by gdyuldin

jflinchbaugh, 2024-10-01:
:heavy_check_mark:  v0.25.0 looks good now. thanks.

commit 3578cedf (2024-09-30): Fix applying filters while changing mode in CAT

Closes: #4