OpenX6100/x6100_gui #5: Feature request. Custom color schemes for eye fatigue during the night and sunlight readability

xes, 2024-09-30:
First of all, thank you for all your efforts!

While I already know that this request should be tagged with the lowest possible priority, I would kindly ask you if it would be possible to manage a custom color scheme for the application.

I really like this app, and I love to spend many hours in front of it during the night, but all this blueish of the gui hurts a little bit my eyes. Sincerely, have the possibility to change some color would help a lot.
And... While speaking about colors, i would suggest also some more color for the s-meter.

This kind of changes would help a lot during the night, but in fact also under direct sunlight to increase screen readability.

Thanks in advance!

gdyuldin, 2024-10-04:
Without complete redesign of whole application I can try to change tint of the buttons and panels background images to some yellow-green, for example. Not sure that I understand you correct about adding more colors to S-meter. Could you provide an example?

xes, 2024-10-17:
Well, on the following page there is a nice comparison of a few graphical interfaces:
https://groups.io/g/xiegu-x6100/topic/99797038#msg3285

I think that some idea could be picked from there. 
For sure, using less blue and some 3d button would help a lot the readability.
Then, in fact if you search for x6100_gui / r1bcu screenshots. almost all of them have the blue buttons and the waterfall completely blue. Really, too much blue... One example:
https://pbs.twimg.com/media/FnrsSdbXEAM491i.jpg

About s-meter, from my point of view, probably the clearest one is the X6100 stock one, but also using more contrasted/saturated greens and red would be enough to make it better on the x6100_gui.

These are just a few ideas, I don't want to distract you from more important fixes and functionalities, but in case in the future you would take a look at the interface, a color theme config would allow a really nice tuning.

gdyuldin, 2024-10-21:
Sorry for the late answer. 
Agree - UI might be better. Also - thanks for UI examples. X6100 is my second transceiver and these examples might be a good point to think about redesign.
I definitely like idea about stacking spectrum and waterfall, will think about it.
Looks like it worth to learn some UI design tools in the future (or find a designer).

73!

W8LID, 2024-10-27:
I also find the amount of blue used to be hard on the eyes. Current UI graphics have a video game menu look to them. I modified my firmware with alternative graphics. I don’t think they’re the greatest but definitely find them easier to look at, particularly in low light. Here’s a comparison of the current UI and my first attempt at alternative graphics. 
[image]

gdyuldin, 2024-10-27:
I guess, as a first step I'll add a parameter to change color of UI elements. Also I going to slightly change color mapping for waterfall (on screenshoot).

Later, I'm going  to move and rework some elements to stack waterfall with spectrum.
[image]

W8LID, 2024-10-27:
A global tint color would be great, it could be used with lv_obj_set_style_img_recolor to tint all the UI images. I’ll give this a try with the firmware that’s included with the buildroot. I have troubles building the current git firmware, buildroot builds fine but struggle with the separate firmware but that’s a separate topic I need to resolve.

gdyuldin, 2024-10-27:
I tried `lv_obj_set_style_img_recolor` - it's looks like overlay of semi-transparent color film. 
I'm going to replace images for buttons, panels, etc with pure lvgl-defined objects, without using any images.
Something like styled button https://docs.lvgl.io/8.3/widgets/core/btn.html?highlight=button#styling-buttons (without shadow).
Also, I'll try to use 2 or 3 base colors from palette https://docs.lvgl.io/8.3/overview/color.html?highlight=color#palette
with `lv_palette_lighten` and `lv_palette_darken` for all objects (except meters, I guess).

Here is early (ugly) prototype with moved blocks from Gimp. Top row - clock, freq, levels. Bottom - buttons. Left side - indicators, perhaps current valcoders action (not sure). Remain area for sprectrum/waterfall. Panels (CW/RTTY decode) and apps (FT8, settings, gps, etc) will be a bit narrower. Frequency and TX power/SWR will not cover central part of the spectrum.
[image]

closed 2024-11-14 by gdyuldin