OpenX6100/x6100_gui #6: Found some bugs

PCBenetton, 2024-10-04:
Hi everybody.

I'm having some issues with 0.24.0, 0.24.1 and 0.25.0. All releases have these same bugs:

- on/off button briefly turns LCD off then resets fw, so no way to turn off LCD.
- battery info and voice mode long press resets fw (seems to be function related and not hardware button related), all other functions work fine, same buttons with different functions work fine.
- all works fine with the "original" 0.20.1 from R1CBU.

Tried different SD cards and flashing with Rufus and Balena. Same results.

I remenber that back with 0.18 and 0.19 releases the lock button soft reset function was not working for me. Instead of the soft reset it was abruptly shutting down the radio. It got fixed on 0.20 for me. Maybe this info helps.

Anyone else having these issues?

gdyuldin, 2024-10-04:
Seems, it's a build order of the buildroot. I changed an action for building image - checking if it will help.

PCBenetton, 2024-10-04:
Do you have the same problem or is it hardware version related?
My SN is V1W#K31368, not sure if it's possible to know which hardware revision mine is. It was bought February 2023.

gdyuldin, 2024-10-04:
No, I have the same issue. Something with initialization voice synth engine. It's not related with application code, but with built OS. If you don't need a voice prompts - you can turn it off in a settings, while I search for solution.

gdyuldin, 2024-10-05:
I created a new release with fix (0.25.1)

closed 2024-10-05 by gdyuldin

PCBenetton, 2024-10-05:
Hey! Thanks! It fixed it!
But I noticed something. The lock button soft reset only works for me if I have external power connected. On the internal battery it abruptly shuts off the radio. And it's the same problem I had back with 0.18 which makes me think that it never got fixed for me, maybe I was just using external power when I tested it.
I do have a battery issue with mine, even at full charge, if I transmit cw and try to tune a few times the radio shuts off abruptly and only turns on after I connect external power (I can hear a relay click and only then I can turn it on). For what I researched it seems one of my cells is bad and can't provide enough current making the BMS shut it all off.
It happens with both xiegu and alternative fw, which further indicates hardware issue.
Maybe this soft reset is causing the same thing? Although when the soft reset turns the radio off I am able to turn it back on without the need for connecting external power.
I have plans to disassemble and try to replace the bad battery cell, any recommendations on with cell to purchase?

gdyuldin, 2024-10-06:
As far as I know - soft reset behaves like an base firmware upgrade - app sends a reset signal to the base board MCU. And as for FW upgrade - if the radio is not connected to external power supply - it just off. 
I tried to reproduce your issue with CW and tuning - my radio wasn't turn off after few session of TX and ATU.
Regarding battery - I have no recommendations. I seen a video with replacement of the accumulators (on russian, sorry) - might be it will helpful.

PCBenetton, 2024-10-06:
Can you send me the link to the video? The auto translated cc on YouTube is decent.

Randy-W0RDR, 2024-10-06:
Thank you for the great software and support. This makes the X6100 even MORE fun to operate.

Спасибо

gdyuldin, 2024-10-07:
https://youtu.be/RY9Dp9rvjl8?t=368
Sorry, I found it, but forgot to paste.

PCBenetton, 2024-10-07:
> https://youtu.be/RY9Dp9rvjl8?t=368 Sorry, I found it, but forgot to paste.

Thanks for the link! Would have never found it myself! It was exactly what I needed! Already ordered the batteries.