OpenX6100/x6100_gui #7: Answering POTA CQ

w4rin, 2024-10-05:
When using FT8 and answering a station calling CQ POTA, it uses POTA as the call sign. 

Awesome work!
Thanks,
W4RIN

gdyuldin, 2024-10-06:
Hi. I know this issue. Try to fix in the next release.

w4rin, 2024-10-06:
Sounds great and thank you for the awesome software!

W8LID, 2024-10-14:
> Hi. I know this issue. Try to fix in the next release.

I fixed this with a simple solution by changing line 1393 of dialog_ft8.c to

`if (strlen(call_de) == 2 || strcmp(call_de, "POTA") == 0) {`

I tested it and it works but I think a more complete solution is needed, found this out when I was testing and came across a SOTA station :). I'm having trouble just building the application and have to do the whole buildroot, my linux skills are rusty, installed it on a machine just to try this out.

[image]

gdyuldin, 2024-10-15:
Thank you for your suggestion. 
I'm going to add a function to check CQ modifiers - such function will be also useful for #8.

w4rin, 2024-10-15:
Awesome, thank you so much!

73,
W4RIN

commit e6970cff (2024-10-21): Add support of incoming CQ modifiers

Closes #7
closed 2024-10-21 by gdyuldin