OpenX6100/x6100_gui #10: Voice announcements broken after tuning with internal tuner

HegrJan, 2024-10-16:
When using the internal tuner the voice announcements stop functioning after a tuning cycle is completed. Simply toggling the ATU on/off doesn't cause any issues, but when long-pressing the ATU button to initiate tuning, the voice announcements stop playing after the tuning beep. The radio remains silent for the duration when announcements are expected. Toggling the voice announcements setting on/off does not restore functionality, and the only way to recover is by resetting the radio.

### Steps to reproduce:

1. Turn on the voice guide.
2. Long-press the ATU button to initiate tuning.
3. After the tuning beep, observe that no voice announcements are heard.

closed 2024-11-14 by gdyuldin

HegrJan, 2024-12-15:
Hello, please, any findings on this? I've noticed the issue is closed as completed. I've tried the last firmware (0.28.0), but didn't spot any changes. Even the recomended work-around with keying in digital mode doesn't work for me.

gdyuldin, 2024-12-16:
Hi Jan. I can't do something with this. This part is controlled by MCU on the BASE board under OEM Xiegu firmware and I can't do anything with this from the application side.