OpenX6100/x6100_gui #22: add direct ft8 encoding and decoding

sysadminpower2019, 2024-10-28:
ft8con and ft8modem exist allowing direct encoding and decoding from a linux cli
https://www.kk5jy.net/ft8con/

since from what i understand the x6100 simply runs linux at it core it should be easy to integrate. 

i understand that memory and disk space might be constraints for that it may be useful to provide a modular system where other components like CW decoding and memories could be disabled / not built to provide the needed / missing space. 

million thanks

gdyuldin, 2024-10-28:
I didn't understand usecase. If you have a PC - you can connect X6100 to it with cable and start using ft8con/ft8modem on your PC without any modifications on the device firmware/program. If you don't have a PC - I not sure, how to communicate with terminal program without keyboard.

sysadminpower2019, 2024-10-29:
thanks for looking into it.

The use case would be to set your grid square locator at home for example
via a keyboard or ssh or some other input method.
Take it out to the field and then just let it run. Creating a truly
portable ft8 station.

I am sure that when it comes to ft8modem and such it can be configured to
auto answer calls and drop if no reply after a while.
I am not sure how the x6100 reads inputs from the radio panel and
microphone but it maybe possible to also set up some kind of input there.

that being said ft8 as far as I understand and how I have operated can be
pretty hands off.
Any changes or setting can be done at home and then the radio packed and
taken on A SOTA expedition without any additional equipment.

gdyuldin, 2024-10-30:
You want a fully automatic FT8 station, which will call CQ and answer all operators without human control of the process? am I right?
Now you can setup QTH with GUI, start calling CQ, and x6100 will automatically answer to first answer and continue QSO with whis operator. You only need to switch back to CQ mode after ending QSO.
Perhaps, it's might worth to add automatical switch back to CQ mode after ending QSO.

sysadminpower2019, 2024-11-01:
yea the ideal would be basically to set up everything before hand and then
work automatically. almost like a beacon. i'm sure this could be adapted to
other digimodes like wspr with the same principal

> You want a fully automatic FT8 station, which will call CQ and answer all
operators without human control of the process? am I right?
> Now you can setup QTH with GUI, start calling CQ, and x6100 will
automatically answer to first answer and continue QSO with whis operator.
You only need to switch back to CQ mode after ending QSO.
> Perhaps, it's might worth to add automatical switch back to CQ mode after
ending QSO.
https://ci3.googleusercontent.com/meips/ADKq_NaNnvxW0wAov1Tac8cY9Mm0onYLdtBHZqZnQ3BS6GmkpAo8qY_fW8XpCVmW5pAMNfcUffVtKLqlKwIkjt6cP18JnJP0ZG-wOjKz497y3D0iBGBTdMbggV9dUmIFbBIvwXZf9uC7NS6xCXo5ayZy3UhqcTbqyp50oSoeVUQO4ZHwUhStYiYCx2tHp61w0CSQqN6oe_nWxADkrpZ6BkOMkvgTZDgnWCImqWUreu_g4Y7sT5YZxtn1wJE=s0-d-e1-ft#https://github.com/notifications/beacon/AJWXABUS72OKPFKRABGQ4X3Z6CLZXA5CNFSM6AAAAABQXJUHFGWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTURZYFWG.gif>Message
ID: ***@***.***>

closed 2025-05-08 by gdyuldin