OpenX6100/x6100_gui #25: Doesnt allow long WIFI passwords

scm2000, 2024-10-29:
Apparently my wifi password is longer than allowed to be typed
The max len allowed for WPA/WPA2 is 63, and the gui only allows you to enter 40

scm2000, 2024-10-30:
thats 63 for alpha numeric and 64 for hexadecimal passwords. 
the problem line appears to be

v_textarea_set_max_length(text, 40);
in textarea_window.c

gdyuldin, 2024-10-30:
Thanks for your report. I'll check.
As a temporal solution you can:
1. connect to radio with usb cable
2. launch `nmtui`
3. select `Activate a connection`
4. select your network and type a password.

GUI application uses the same library for wifi connection (`libnm`). After successful connection with `nmtui` connection parameters will be saved on the system and you will be able to use this saved connection within GUI application without need to provide a password.

scm2000, 2024-10-30:
Thanks.  I'll need a bit more help though... 
I assume you mean connect my computer to the dev usb port, and use a terminal emulator...  which of the two com ports, and what baud?
If not that, what?
Thanks again.

gdyuldin, 2024-10-30:
Yes. You can use PuTTY under windows.
Baud is 115200. Not sure about correct port number, but you can try both. After connecting - press enter. It should ask a password, if not - try another COM port. Login is root, password is 123.

scm2000, 2024-10-30:
> Yes. You can use PuTTY under windows. Baud is 115200. Not sure about correct port number, but you can try both. After connecting - press enter. It should ask a password, if not - try another COM port. Login is root, password is 123.

That all worked, thanks!

commit 3904fafc (2024-11-02): Allow passwords up to 64 characters

Closes #25
closed 2024-11-02 by gdyuldin