OpenX6100/x6100_gui #29: WIFI password will not accept capital letter as first character

packinpastor, 2024-11-03:
When trying to input the WiFi password, I cannot get passed the first letter. If I input a capital letter for the first letter it gets stuck and will not take any further input. If I type a lower case letter as the first letter then it takes the input without getting stuck on the first letter. I used the nmui work around via putty to get the password input and radio is now up on WiFi.

gdyuldin, 2024-11-04:
Hi. Using on-screen keyboard I can't reproduce it. 
Unfortunately, I have no USB keyboard to check.

packinpastor, 2024-11-04:
I don’t get an on screen keyboard when inputting the password. A USB keyboard is the only option. Should I be getting an onscreen keyboard when inputting the WiFi password? I don’t

gdyuldin, 2024-11-04:
I'll try to check with USB keyboard. Thanks for your report.

gdyuldin, 2024-12-04:
Confirmed. Something strange with capital letters at any position.

commit 1540d326 (2025-02-03): Fix external keyboard upper case

Closes #29
closed 2025-02-03 by gdyuldin