OpenX6100/x6100_gui #39: Band editor for FT8/FT4

gdyuldin, 2024-11-07:


BadamK, 2024-11-08:
Please show the Frequency being used for FT4/FT8.

and allow to change frequency,... (There are other frequencies used for other events)

gdyuldin, 2024-11-10:
I'm going to move some elements of the UI - so frequency should be always visible.

AvidFlyr, 2024-11-14:
To expand on BadamKs post, i.e. display and QRG change: currently, band changes of the main VFO are not carried over to the FT8 module. FTx as a simple mode change (as in CW/SSB/FTx) should be emulated. To illustrate: if you were last on 20m FT8, are now on 40m SSB 7074 to check the activity and now hit APP/FT8 it's irritating find yourself back on 20 again.
Perhaps make this an option in settings ("FT8 band from VFO band y/n")?

gdyuldin, 2024-12-03:
Available on the last release with webserver  - https://github.com/gdyuldin/x6100_gui/wiki/webserver

closed 2024-12-03 by gdyuldin