OpenX6100/x6100_gui #41: Allow POWER Setting selectable for MFK

BadamK, 2024-11-08:


gdyuldin, 2024-11-10:
I'm not sure about how to do it better. Right now settings "tied" to vol or mfk encoder in the code. Sometimes I forget about defined mapping and rotating wrong encoder. I might leave only volume on VOL encoder, move all other things to MFK, not sure.

PCBenetton, 2024-11-12:
Maybe holding down the function first assignes it to the VOL encoder, then hold again for MFK, and hold again to clear. And in the GEN menu you can remove the VOL MFK labeling and change it to something like "Encoder" just to be clear on what functions are possible to link to either encoder. Or maybe just a color code and that opens up the first soft button to house a function and not a label, reducing a bit the number of "pages" needed.
Or even an approach more like the stock fw where the whole screen gets filed with a grid of options, reducing pages, and the assignment is the same (hold 1 for VOL, 2nd hold for MFK, 3rd hold clears).
73!

gdyuldin, 2024-11-12:
I'll think about it. Anyway - changing assigning logic requires some refactoring - so it won't be a first priority task.

AvidFlyr, 2024-11-16:
> I'm not sure about how to do it better. Right now settings "tied" to vol or mfk encoder in the code. Sometimes I forget about defined mapping and rotating wrong encoder. I might leave only volume on VOL encoder, move all other things to MFK, not sure.

The principle of splitting the functions to the two encoders is fine to avoid overload on the MFK encoder.
However, the vol encoder should be restricted to:
1. AF gain (default)
2. RF gain 
3. Filter bandwith (bw much better for CW than complicated upper / lower settings). 
Candidates only - these are not changed all the time and accessing the via a FN key is fine: 
- zoom level
- tx PWR

gdyuldin, 2024-11-17:
The assignments is saved in the database - so it not lost after reboot.
You can check how to do it in the wiki - https://github.com/gdyuldin/x6100_gui/wiki#front-controls
I'm still working on the documentation, but I hope it might be helpful.

closed 2025-05-08 by gdyuldin