OpenX6100/x6100_gui #45: Band up / down switch - no 60m

AvidFlyr, 2024-11-17:
Currently band up / down generates 160m, 80m, 40m x2 (7035/7044), 30m, 20m x2 (14030/14060), 17m, 15m, 12m, 4x 26-27 MHz, 10m and 6m.
No 60m
Ideally, moving up and down should place the VFO on the last frequency used (1 per band) of all ham bands. User-editable config file to add CB band if required.

gdyuldin, 2024-11-17:
I'm going to add band plan editor - you will be able to add/edit/remove bands.
Saving/restoring VFO for band is already implemented.

zerog2k, 2024-12-03:
I really love how the band info (e.g. `40m SSB` or `41m Broadcast`) is overlayed onto the waterfall when tuning. This is a great feature.

I also wanted to edit the band plan, as I'm in ITU Region 2, and we have a different band plan from what is in the radio currently.

Suggestion, I see the band plan is already imported into sqlite db from a csv file.
https://github.com/gdyuldin/x6100_gui/blob/main/sql/bands_ham.csv
Maybe this can be just a csv file that sits on the root of the sdcard, and which is imported/overwritten into the sqlitedb on every radio startup - so that user can select preset ITU region band plan csv, or edit their own file, and then move the chosen file into place on the sdcard (e.g. band_plan.csv)

gdyuldin, 2024-12-03:
You can configure your bands with webserver (from last release) - https://github.com/gdyuldin/x6100_gui/wiki/webserver

closed 2024-12-03 by gdyuldin