OpenX6100/x6100_gui #46: Constant "transmit" spike on selected frequency showing on waterfall - CB, SSB.

marcbirkett, 2024-11-17:
Hi,

I'm doing well on FT8 but whatever frequency I select I get a constant transmit spike showing on the waterfall as per the picture. Any way to get rid? It makes the waterfall difficult to use. Otherwise, enjoying it very much, good work!
[image]

Ta

Marc

gdyuldin, 2024-11-17:
Application uses I and Q signals from the BASE board to calculate spectrum and waterfall. Could you check it:
* without connected power (using internal battery)? Perhaps, it's a noise from power line.
* with turned off wifi module. It also might be a source of noise.
* with turned off display (short press on power button). Then, after turning on - check the difference between waterfall.
* using original firmware (Just turhn radio off, pull SD card and boot)

marcbirkett, 2024-11-18:
hi, thanks for the reply.

1 - it is still present when on internal battery
2 - it is still present with the wifi turned off
3 - makes no difference
4 - it is present on original firmware, but much, much smaller as per attached.

[image]

Commonly the waterfall "freezes" too and I have to reboot the machine, i'm adjusting the settings to look at that too.

Thanks for your input.

Ta

Marc

marcbirkett, 2024-11-27:
Error logs below, spike and waterfall freezes requiring restart.

[x6100_log_2024-11-27_13-52-36.txt](https://github.com/user-attachments/files/17935571/x6100_log_2024-11-27_13-52-36.txt)
[x6100_log_2024-11-27_13-54-56.txt](https://github.com/user-attachments/files/17935572/x6100_log_2024-11-27_13-54-56.txt)

marcbirkett, 2024-11-27:
I can, testing now.

marcbirkett, 2024-11-27:
hi, that has solved the waterfall locking up. Awesome! The spike on the selected frequency remains, but nothing in the log about it

gdyuldin, 2024-11-27:
As far as I know - xiegu made some different HW versions. It might be a reason. You could try to flash a different version of the BASE firmware, but I'm not sure, that it will help.

marcbirkett, 2024-11-27:
HW? I'm on the most recent firmware at the moment, will give it a try

marcbirkett, 2024-11-27:
tried 4 different base units, all pretty unusable for SSB as per picture. never mind, made contacts on FT8. Will keep upgrading firmware as it becomes available. Thanks for your efforts, this can be closed.
[image]

closed 2024-11-28 by gdyuldin