OpenX6100/x6100_gui #48: Unexpected restarts on FT-8

marcbirkett, 2024-11-25:
When using FT8 on 20m restarts at seemingly random intervals without cycling the power. I.e. the power is still on but it "looks" like its just been booted up i.e. reverts to 40m SSB and any saved settings are lost.

gdyuldin, 2024-11-25:
Looks like a bug. Perhaps, something useful might be in the logs. Could you share one of them (it's within DATA partition, under app_logs directory)

marcbirkett, 2024-11-25:
[x6100_log_2024-11-16_14-44-31.txt](https://github.com/user-attachments/files/17907079/x6100_log_2024-11-16_14-44-31.txt)
[x6100_log_2024-11-16_14-49-19.txt](https://github.com/user-attachments/files/17907080/x6100_log_2024-11-16_14-49-19.txt)
[x6100_log_2024-11-16_14-51-18.txt](https://github.com/user-attachments/files/17907081/x6100_log_2024-11-16_14-51-18.txt)
[x6100_log_2024-11-16_15-21-45.txt](https://github.com/user-attachments/files/17907082/x6100_log_2024-11-16_15-21-45.txt)
[x6100_log_2024-11-16_15-54-18.txt](https://github.com/user-attachments/files/17907083/x6100_log_2024-11-16_15-54-18.txt)

marcbirkett, 2024-11-25:
everything in the log directory above, was expecting to see something with yesterdays date, so will attempt to replicate this week and post anything new. Lots of things going on though!

marcbirkett, 2024-11-27:
Can replicate when doing an SWR scan on my attenna, this happens about half the time. Change band to 20m (in  my case 14.073), then:

1. Tune - atu tunes attenna
2. App
3. SWR Scan
4. Run - scan runs
5. Run to stop the scan
6. App - radio resets to 40m

Continuing to test with FT8

marcbirkett, 2024-11-27:
Also, FT8 on 20m.

1. Send CQ
2. App to stop FT8
3. Radio resets

marcbirkett, 2024-11-27:
but no further logs created in app_logs

marcbirkett, 2024-11-27:
Also, on FT8, 20m. Reply to someone elses CQ, get a reply from them, on next transmit cycle, radio resets

gdyuldin, 2024-11-27:
I didn't finish logging, right now it have no controls. Also I can't reproduce on my radio.

I can build a debug version of the application for investigation.

You will need to:
* Copy application to the sd card
* Connect to the radio with SSH (if your radio is connected to the WIFI) or with PyTTY (using USB cable)
* Stop original application, launch debug version (I'll share all commands)
* Try to reproduce issue
* Share output.

Is it possible?

marcbirkett, 2024-11-27:
Certainly, however, i've think i've made progress. I've started from the beginning, formatted SD card, downloaded latest 0.27, and recreated from scratch, and the restarting seems to have stopped. I'm currently trying for a FT8 QSO then going to check the error logs. Will come back to you.

marcbirkett, 2024-11-27:
So, resets stop, and now I have error logging back again. This one can be closed. Ta!