OpenX6100/x6100_gui #51: Extend Transverter Frequency Range to include 4m band

infrat, 2024-11-27:
Hi fellow hams!

The current firmware sets the lower limit of the transverter frequency range at 100 MHz, which excludes operators in regions where the 4m band (around 70 MHz) is available to use their TRVs. This limitation affects the usability of the transverter feature for those wishing to access this band.

I've created a very small change allowing user to set lower limit in transverter settings.

https://github.com/gdyuldin/x6100_gui/pull/50

gdyuldin, 2024-11-27:
Hi.
Thanks, I merged your PR.

closed 2024-11-27 by gdyuldin

infrat, 2024-11-27:
@gdyuldin I forgot to extend transverter shift range in my previous pull request. 

In new PR I've also added new frequencies to DB. 
https://github.com/gdyuldin/x6100_gui/pull/53 
now it works smoothly with my 70MHz TRV.
Vy 73 de SP9HGN

gdyuldin, 2024-11-28:
Done