OpenX6100/x6100_gui #55: Label button on screen for "exit" or something that would RETURN to previous screen,...

BadamK, 2024-12-01:
Too many screens don't have a clear indicator or HOW to exit and return to previous screen,...

Or screens are Inconsistent with what control to exit with

gdyuldin, 2024-12-02:
Hi. Long press on the 1st button below the screen loads previous screen.

BadamK, 2024-12-09:
That doesn't work on all screens,... And I think one screen I find trouble with is when the keyboard is displayed,... Or maybe when entering a screen and then not actually wanting to make any change,... Or maybe entering the settings page,... I've just seem to be somewhere in the system and have to press the GEN or APP button to get out of it and not be back to where I thought I should be,...then forget where I was trying to make changes,...

If I remember to I'll try to write down the problem screens,... If none else is noticing these issues, feel free to close out this issue. 

 If there is any sort of manual , a link to it might help me .   I don't use the firmware that offen to keep up with the changes.  

It's nice to see "working"  apps,...FT8 and FT4,... Thanks for FT4,...
  I've attempted decoding RTTY and CW and BPSK that are provided with the Xiegu firmware and feel like they are too difficult to figure out,... I don't find much to decode,... 
It's likely I don't understand the modes that well to know how, and where to look for this activity..