OpenX6100/x6100_gui #59: Time update from WWV or similar,...

BadamK, 2024-12-04:
One thing I have now seen anyone do yet,.. except for the atomic clock from MFJ

And I'm surprised that it is NOT an option manufacturers have attempted to utilize,...

Although not so many radios have built in apps that require time sync,... 

But the also, why not make it part of a software package,...

The ability to just tune into one of several frequencies already broadcasting the current time to the second, and with the users timezone, be able to update the time via that option,... NO Internet required.

Seems like a cool idea to me,... And if the reception can't be received then have the usual options,... Or just wait until it can be received,... ..

Might take more user interaction to cycle thru all the different frequencies,  any make sure they have the right antenna setup for each and tuned,... But this is HAM radio,...let's try and use the radio part as much as we can,...

Thanks.

gdyuldin, 2024-12-04:
It's interesting task, but after some searching I can't find any WWV signal decoding library. Implementing this feature from the scratch may requires many efforts.
Let it be here, but it will a low priority task.

BadamK, 2024-12-07:
I'm not sure if there are other time based beacons being transmited around the world for others to receive,... Some of the propagation maps I've seen seem to suggest that it might only be received best around North America.

BadamK, 2024-12-07:
And I'm not sure the coded signal with time info is only transmited on 60KHz,.... And not on the voice frequencies of 2.5, 5, 10, 15, 20, 25MHz.

I've never looked that deeply into the broadcast of the signal, ..  just know we have one clock that syncs with the signal,.. and I've heard the broadcast on the other frequencies,.   Don't currently have a receiver for 60khz to know what is there.

zerog2k, 2024-12-07:
Yes, on WWV/WWVH there is a coded signal on a 100hz sub-carrier.
https://en.wikipedia.org/wiki/WWV_(radio_station)#Digital_time_code

zerog2k, 2024-12-07:
and there may even be WWV/WWVH time decoding/demodulating software drivers already available in ntp:
https://www.ntp.org/documentation/drivers/driver36/

zerog2k, 2024-12-07:
with all of that said, it may be just easier to plug in a cheap usb gps "mouse" to get NMEA serial GPS (and therefore time) data.

I think it would also be curious if we could also get (and set) the grid locator data automatically from the GPS coordinate data.

This should be straightforward to convert gps lat/lon to maidenhead
https://github.com/sp6q/maidenhead

gdyuldin, 2024-12-08:
Time and grid locator should automatically synchronized from GPS.

closed 2025-01-10 by BadamK