OpenX6100/x6100_gui #67: Saved WiFi

TheDireHavok, 2024-12-08:
Will not connect to saved WiFi after 1st connect. Connected successfully to WiFi 1st attempt. But when trying to connect after WiFi is turned off connect button does nothing when selecting saved SSID.

TheDireHavok, 2024-12-08:
Added issue: tried to connect after 1st success to a secondary SSID, it allows password input and upon complete(selecting checkmark) it goes back to Scan List and it never connects. selected connect to same SSID and asks for password again. Unable to connect to any SSID again.

gdyuldin, 2024-12-08:
Could you try to delete saved network and try to connect it again?
Could you share application log (from `app_logs` directory of the SD card)?

TheDireHavok, 2024-12-08:
I Deleted saved SSID, and tried to connect again to successful SSID, but no luck. will send log

TheDireHavok, 2024-12-08:
app_logs folder is empty

TheDireHavok, 2024-12-08:
gonna try older version of FW on another card to see if its same

TheDireHavok, 2024-12-08:
NM there isn't wifi on older FW, noticed too no QTH on new FW

gdyuldin, 2024-12-08:
Sorry, I have no enough information to understand your problem.
Could you:
* burn new firmware to sd card
* reproduce your issue
* describe all your steps, actual results and expected results for each steps

Also, don't understand about QTH - what goes wrong? Is it related with WiFi? If no - please open different issue.

closed 2025-05-08 by gdyuldin