OpenX6100/x6100_gui #89: LOG view in Webserver-Interface without downloading logfile

dk5cr, 2025-01-02:
Would it be possible, to add a button with logview and some other functions to edit certain log entries, delete or correct them if necessary? 
Also functionality like clear log or clear parts of the log (before YYYY-MM-DD or logentries of a certain type like ft4) would be very helpful... 

Thanks for great work on this project. 
73 de Cornel DK5CR

gdyuldin, 2025-01-02:
I'm not sure, that I understand use case. Is there some errors in the ADI file, generated by transceiver? Also, editing this file will not impact anything on the radio - it only write qso to this file, never read information from it. I guess, if you want to modify something within ADI file - it will be better to use one of the exists editors for PC (like ADIF Master)

dk5cr, 2025-01-03:
Thanks for your reply. It's like you have a Logbook on your PC, perhaps you have dupes or so, you wanna delete. In the current situation you have to download the adi-file and correct the entries, so it would be nice to have an "online"-overview to the log via Webserver

gati111, 2025-01-04:
Фантазирую. Вот если бы  еще добавить возможность вести лог (аппаратный журнал) не только в FT8, это было бы бомба! Пристроить самый простейший лог что бы записать корреспондента QSO, а Log  мог считывать частоту, моду, время и сохранять это на SD. Было бы WOW! Что бы одной кнопкой переключаться между логом и мордой, а лог открывался бы сразу с клавиатурой. При реализации такой возможности отпала бы необходимость таскать с собой в поход всякие планшеты ноутбуки и прочее, или вести бумажный лог. Я вот иногда забываю выставлять частоту проведенного QSO и потом не могу вспомнить. По большому счету этот трансивер и создан для походов и выездов, а в рюкзаке место всегда в дефиците.

BadamK, 2025-01-20:
Maybe someone would like to enter additional NOTES to a log entry.,... Type of antenna used,... Location notes , .. or something else,...

dh8iat, 2025-03-16:
The FT8/FT4 app is a truly great development. I really like it. A way to delete all QSOs in the log via the webserver would be a great feature.
73 de DH8IAT, Thomas