OpenX6100/x6100_gui #92: QSO problems with Non-standard callsigns

el-musafir, 2025-01-11:
Hi Georgy, first of all many thanks for this most excellent software which I have been using now for two months with excellent results. The only issue I had was that sometimes when calling a station with non-standard callsign like “II8RCOS” in ft8/4 the second part of the QSO with the report is not sent - the rig does not switch to send, you can hear the audio though. This also happened to me with stations having “slash” callsigns, maybe it is something with the underlying ft8lib? Best, Eric DF3TN

gdyuldin, 2025-01-16:
Thanks for your report. I pushed some changes for non-standart callsigns to https://github.com/gdyuldin/ft8_lib - it might help. This changes should be included to the next release. I leave this issue opened to check again later.

el-musafir, 2025-01-16:
Thanks for all your great work. I would enlist as a tester for these fixes as soon as they would be available. I’ll try to build the image one of these days next week anyways. Best Eric, DF3TN

gdyuldin, 2025-02-16:
Should be fixed in https://github.com/gdyuldin/x6100_gui/releases/tag/v0.29.0

closed 2025-02-16 by gdyuldin

el-musafir, 2025-02-18:
The issue is fixed in 0.29.0 - perfect work, thank you very much!