OpenX6100/x6100_gui #94: Manual time sync for FT8/FT4

BadamK, 2025-01-15:
I noticed the Time Sync button on Page of FT8/4 menu,...

Can this button sync FT8/4 time instantly when the operator presses it?

Use case,... While listening to the END of an FT8/4 transmission,  during the SILENCE period in between the next tx...press time sync,... 

This would be more useful,... 

ADVANTAGE,...
FT8 time does not have to sync on the "minute",  to work someone that is not synced to the minute start.

Flexibility to sync within a reasonable second or two ?

DISADVANTAGE,...
 Log entry TIME might  not be in sync with ACTUAL time

BadamK, 2025-01-15:
Maybe this is how it already operates,...

I just checked if I could get OUT OF SYNC by pressing time sync ,.. and it did seem to stop decoding,... When I pressed time sync again at the end of the FT4/8 rx,... It seemed to start decoding properly again.

Feel free to close this issue.

gdyuldin, 2025-01-16:
You're right - pressing time sync adjusting clock to closest time slot.

closed 2025-01-16 by gdyuldin