OpenX6100/x6100_gui #98: CW decoder stops working after many hours

jflinchbaugh, 2025-01-31:
If I'm decoding CW, and I leave the radio overnight with the decoder running, the next day, it's unable to decode anything.

Things I've tried that don't fix it:
* clearing the screen by pressing GEN
* tuning around to other signals on the same band
* switching to an SSB mode and back to CW

Things I've tried that do fix it:
* turning the radio off and back on (of course)
* long-pressing the lock and restarting the UI

I'm sure there are other things I can try next time I see it happen.

gdyuldin, 2025-02-01:
You can try to use a recorder app to record and listen what signal comes to the main board from base.
Also - settings app has a Play/rec gain - last is a control for audio level comes to main board.

jflinchbaugh, 2025-05-05:
Another observation: I see the screen eventually clear the decoding dialog, and I'll  see it decode and write a single character for a moment, and it clears or scrolls away to be a clear screen again.

gdyuldin, 2025-07-01:
I leave x6100 with CW decoder turned on for the last night - nothing happens.
If you catch this issue again - could you try to record a voice message (switch to ssb, open MSG and press REC on 2nd page) and record RX sound with recorder application (2nd page of APP).
Then - try to play both recorded files.