OpenX6100/x6100_gui #101: FT8/4 mode: volume/power knob without unwanted function on new libft8.so

dk5cr, 2025-02-07:
Hi, 
just tried again the new ft8lib.so file on the x6100. Unfortunately sound volume has no effect in ft8 mode. Even if volume is set to 0, the ft8-send sound is to hear from the loudspeaker, the received audio remains quiet. Pushing the volume knob to change between power and volume ends the ft8 mode and switches back to "normal" mode in old band setting (e.g. 20m), even if i changed band in Ft8 to 40m. 

Ft8 CQ is now working and is shown on my second device, so this issue (#100 ) seems to be solved with new ft8lib.

gdyuldin, 2025-02-07:
Hi. Pressing on the VOL encoder within any app acted as escape key. I guess Oleg R1CBU did it to prevent some unexpected changes in the radio state, when application is active, or to reduce notifications (which appears, when you turn VOL knob). 

I can't reproduce issue with no effect of turning vol knob on my radio. Let's wait for the next release - maybe I already solved it.

samhiatt, 2025-02-15:
I noticed that (on v0.18.0) I can change the FT8 TX sound by changing "Moni level" on the "VOL 3:4" page, but that I have to exit the FT8 app to select that param first. Interestingly, once that is selected I can still change the monitor level with the vol encoder after entering the FT8 app (although there is no visual indication of what parameter it is changing when I turn it while in the FT8 app).

Obviously it'd be nice to be able to change these parameters (plus RF gain and TX power) while inside the FT8 app, but I haven't written this into a formal feature request yet. It seems like you're moving faster than I can write those. ;)

gdyuldin, 2025-02-16:
Moni level is just a self-control option on the BASE board. If it set to non zero - you will hear sound of your transmission (before it will be filtered and prepared for modulating carrier). In case of SSB - you'll hear your voice. More value - louder monitor sound. Volume control also affect monitor volume.
But monitor level is not affects TX modulation signal level.

BadamK, 2025-02-17:
Would rather be able to adjust volume and power with knob in FT8/4 rather than "esc" out of app,...

We can leave the apps with APP OR GEN key.

V 0.29.0

closed 2026-02-18 by gdyuldin