OpenX6100/x6100_gui #104: 0.29.0: CAT setting mode does nothing

jflinchbaugh, 2025-02-16:
I just loaded 0.29.0. I'm reporting a regression from 0.28.0. it used to work in the old version.

Setting the mode from `flrig` used to work fine, but now it does nothing with the radio. 

I change it in `flrig`, the radio doesn't change, then `flrig` reads mode back to bounce back to whatever the radio previously had.

workaround: 
Changing the mode on the radio works fine, and `flrig` sees the change of mode.

Note, `flrig` is reading USB-D and LSB-D as USB and LSB respectively, not as their digital modes, but the radio stays set to digital mode just fine.

gdyuldin, 2025-02-17:
Fixed, thanks for report. Will be in the next release.

closed 2025-02-17 by gdyuldin

gdyuldin, 2025-02-18:
My bad, will fix it in 0.29.1 (hope, on this week)

gdyuldin, 2025-02-20:
On my dev machine was installed previous version of flrig, without x6100 support, so I tested it with x5105 profile.
Now I updated version of the flrig and reproduced issue. 
Seems, also digital modes get/set will be fixed.

jflinchbaugh, 2025-02-23:
This is looking good now on 0.29.2. ✅

jflinchbaugh, 2025-02-23:
I also noticed an issue yesterday with 0.29.1 where WSJT-X controlling the X6100 by itself (no flrig) would keep switching the mode to USB from USB-D, and then the radio wasn't accepting audio from the sound card. 0.29.2 fixed this as well. ✅

commit f3ad5c45 (2025-02-17): Fix set mode using CAT

Closes #104