OpenX6100/x6100_gui #105: 0.29.0: SWR Scan leaves VFO freq mismatching displayed freq

jflinchbaugh, 2025-02-17:
Reproduction:
1. tune to 28.074 or another FT8 freq where you can see obvious signals
2. observe ft8 signals in band
3. start SWR Scan
4. stop swr scan with the "current" sweep frequency in the first half of the range (below 28.074)
5. press GEN to exit SWR Scan
6. observe that the freq still says 28.074, but those expected ft8 signals are no longer in the listening band

Workaround:
- turn the VFO knob to shift the displayed frequency up and back down, and the "active" frequency will line up again with the displayed frequency.

This initially appeared to be a "deafening" of the radio after SWR Scan, but then I realized I saw other CW signals, and eventually my FT8 signals in unexpected places on the waterfall depending where I randomly stopped the SWR scan.

Extra info that may not be related:
This may also have interacted badly with the ATU and transmit, since I'd then get a low SWR (< 1.5), and lower power (~2W), but ALC would be 7.6. Turning the the radio input signal (USB-D) from `wsjtx` on the computer down to 0 didn't effect the ALC or power at all, so it was like it was getting a signal from somewhere else when in this bad state.

gdyuldin, 2025-02-17:
Thanks for clear and detailed report.
A significant part of changes for v0.29.0 is adding observer pattern for current radio state. Different parts of application might subscribe to changes of the some values (e.g. foreground frequency). This way I don't need to call a bunch of function on every place in the application where radio state changes. Disadvantage of this approach is that every change of state notify all subscribed components - it might be a bit longer. SWR scan application is not using foreground frequency subject, but directly changes frequency of the radio for performance reason. I guess, I miss restoring frequency according current state on the exit of the SWR app. I'll check it in a few days.

Regarding ALC, SWR and power - all these values are received from the BASE board, GUI app just displays them. With digital mode radio "listen" for input signals from the MAIN board and from the USB soundcard. Could you try to disconnect USB and try to push PTT on digilat mode to be sure, that this signal is not generated by MAIN board?

closed 2025-02-19 by gdyuldin

jflinchbaugh, 2025-02-19:
I just tried 0.29.1, and the current frequency is restored fine upon exiting SWR Scan. ✅  I have a reproduction for the second issue, and I'll open a new issue for that one.

commit 120addda (2025-02-19): Fix restoring freq on exiting SWR scan app

Also, fix crash on exiting while SWR is running

Closes #105