OpenX6100/x6100_gui #106: V 29 No power output when using WSJT-X after enabling ATU

PD0RZH, 2025-02-18:
Version V29 Beta
Using WSJT-X for FT8
FT8 works flawless when not using the ATU
After using the ATU no power output

renatoflp, 2025-02-18:
I managed to work around the problem by opening the FT8 APP on the x6100 and controlling it through JTDX (actually FLrig).
In this case, even with the ATU activated, the radio receives audio signal and sends power to the antenna.

gdyuldin, 2025-02-18:
Hi. After finishing ATU indicator should switch to green, than again red for a short time. If there is no short red blink - perhaps, I set too small delay.
Could you check it?
Also, as a workaround you can try to switch your radio to USB-D (not sure, but might help).
Or open FT8 app after ATU, do a short transmission (CQ), then stop and exit app (should help).

renatoflp, 2025-02-18:
Hello.
The red LED flashes briefly until the ATU is tuned.

As for the USB-D, I can't set it via the radio outside the FT8 APP, or even via FLrig, I switch to USB-D, and it returns to USB again.

I hope this information can help.

PD0RZH, 2025-02-18:
> Hi. After finishing ATU indicator should switch to green, than again red for a short time. If there is no short red blink - perhaps, I set too small delay. Could you check it? 
_Led's blinkin' fine_

> Also, as a workaround you can try to switch your radio to USB-D (not sure, but might help).
_Radio was already in USB-D mode_

> Or open FT8 app after ATU, do a short transmission (CQ), then stop and exit app (should help).
_This is a workable workaround_

closed 2025-02-18 by PD0RZH