OpenX6100/x6100_gui #107: SWR Scan causes distortion in RX and some extra signal for TX

jflinchbaugh, 2025-02-19:
I found this while investigating #105. Initially observed in 0.29.0, but also found in 0.29.1.

Reproduction:
0. No USB Connected. Power happens to be set to 8.8, but I don't think it's important.
1. Tune to 28.074MHz (FT8), USB-D.
2. Run the ATU to get a tune.
3. Press PTT for a couple seconds, and observe no power, no SWR, no ALC,
4. Observe a normal FT8 signal and how it sounds.
5. Run SWR Scan, and exit it. (GEN)
6. Observe the FT8 signal is now louder and distorted.
7. Press PTT for a couple seconds, and observe the power meter says it's TXing now and ALC and SWR are high.

Work-arounds to restore normal operation:
- Switch mode to LSB-D and back to USB-D.
- or run the ATU.

I think there's some other state not being restored after the SWR Scan stops.

gdyuldin, 2025-02-20:
Radio remains on AM mode, which is used for SWR scan. Will fix in the next release.

BadamK, 2025-02-20:
Thanks for the continued support of the project,...
And just confirmed that issue is present in 0.29.1

closed 2025-02-20 by gdyuldin

jflinchbaugh, 2025-02-23:
This looks good now in 0.29.2. ✅

commit 97b50523 (2025-02-20): Use subject to mode change for SWR scan

Closes #107