OpenX6100/x6100_gui #127: DNF AUTO. filter sensitivity and duration controls,...

BadamK, 2025-02-23:
I finally had an instance to use the auto function on LSB.

I slow moving signal was passing thru the audio spectrum from high to low and back slowly moving high again.

Auto DNF worked a little better to find the signal at higher tones.  At middle and low tones it had a hard time.  A had to revert to manual control.

Some sort of Sensitivity adjustment might be helpful for users. And maybe a Duration adjustment also, to fine tune how long a signal should be present before the auto  feature acts on the tone.  

I'll have to look back and see if you mentioned what parameters is used to detect a signal.

gdyuldin, 2025-02-26:
The tone detector has better sensitivity on the high freq. With same sensitivity on all frequencies it tends to reject strongest frequency from the operator voice.
You can take a look here - https://github.com/gdyuldin/x6100_gui/blob/main/src/dsp.cpp#L310

BadamK, 2025-02-27:
I don't think we need to worry about voice tones,... The main tones we mostly try to filter is constant tune up tones or birdies that float around the audio spectrum ,... Which are either stationary or move slowly,... But interfere in a random spot.   

We are looking to narrow to absolutely minimum witch on these signals to cancel them out,... These tones are not very wide,... And a variable set time to detect them should be helpful,... 

Keep the detection level adjustable for the operator for the full frequency range.  And having an adjustment for cancelling width we can control also.

If someone is looking to filter voice tones,... A multi band frequency equalizer should be created .

I looked thru that file wondering which values, if they were adjustable, could help us fine tune the auto tone filter.

I definitely noticed when the auto filter hit mid and low tones, it stopped working altogether.  I would rather see it detect the same thru out the whole range, and narrow the detection width. And adjustable amount of time to detect a tone would be helpful i think.

I think I've heard some radios can even had a choice to cancel out more than one tone.

Thanks for the info about this,... 

Line#s with entries I has wondering about in file you linked to:( what would happen if the value was adjustable?) (not necessarily in order,... And not full comments,... It was basically anything that had a number in it,...)

298.   Size_t peak_width = 150 
300.   / 2
332.  Mean_freq / 50 * 50

372.    Psd_delay = 4
388.    Psd_delay = 4
563.    Psd_delay = 1

422 & 436.    -30.0f

424.     -1.0f) / 2.0f + 1.0f
452.     -121
458.   0.8f
490.     Min_max_delay = 2

311.   5000.0f + 2000.0f * 6.0f

520 & 522.      8, 60.0f
 
589.   32768.0f

613.    Size * 15 / 100
614.     Size * 10 / 100

622 & 623.   Min > S8
625.     + 48

643 & 645     (0.4f)

That was probably too many lines to look at,....
I don't know the code so I was curious a little,...

closed 2025-06-03 by gdyuldin