OpenX6100/x6100_gui #132: CW Decoder Popup/Layer Persistent Issue

krazakarl, 2025-03-07:
## Expected Behavior
CW Decoder popup layer should be disabled after tuning out of a defined CW band.

## Current Behavior
Going from a band with CW mode defined, and then entering a pre-existing Broadcast band with an undefined mode keeps the CW decoder popup/layer enabled.

## Steps to Reproduce
1. Tune to a band with a defined CW mode
2. Tune to a band with no defined mode, Example: Default Broadcast bands with no defined mode.
3. Note CW decoder doesnt disappear tuning into Broadcast band.
4. Tune to a band with a non CW defined Band. Example: 20m with USB defined.
5. Note CW decoder disappears after entering a defined non CW Mode.

krazakarl, 2025-03-07:
After discussing this with my colleagues, this may not be a bug, but a feature request.

Scrolling through a band with CW defined, and going into a band that is undefined, the assumption could be made that the radio stays in CW mode.

But for someone who mainly operates in SSB for example, scrolling through CW and it not going back into SSB might be a sub optimal user experience. 

Suggestion would be to potentially storing the previous mode in memory before going into CW. If the Radio was turned on in CW mode, the last mode stored would be CW.

gdyuldin, 2025-03-08:
In terms of the software bands is a "storages" for some settings (mode, att, pre, etc).
If you leave band - you settings remains the same as before.
If you tune/switch to band - settings from this band is loaded.

If you prefer ssb - you can switch to CW band, set SSB and it will store this mode for band.

EmrysTheWzrd, 2025-03-08:
Would it be possible to save the last used settings (mode, att, pre, etc) for the radio as a second array of variables when the settings are changed and the radio is outside of any bands defined in the band editor?

When the user leaves band or enters a frequency that is not in the band editor, the radio would return to the last used settings from the second array. This would assist a shortwave listener who may be tuning between large ranges on the spectrum.

gdyuldin, 2025-03-10:
Not sure I'm understand your correctly.

gdyuldin, 2025-03-12:
Understand. Will think about it.