OpenX6100/x6100_gui #134: ATU boundaries are very close

jflinchbaugh, 2025-03-14:
I may have read in a commit log or release notes that you did something different with the boundaries between where ATU settings are saved and applied, and I think I'm seeing the ATU settings not applying as "widely" as they could, so tuning in one spot for FT8 operation doesn't apply to the TX when WSJT-X shifts frequency just 1kHz, so TX (mysteriously) has very high SWR.

## Expected Behavior
I can set the radio to 7074kHz, activate the ATU, and have it give good SWR within a couple kHz in each direction.

## Current Behavior
1. I set the radio to 7074kHz
2. activate ATU
3. check SWR scan, and it looks nice and low
4. ensure WSXT-X has split operation set to "fake it"
5. TX from WSJT-X on the high side of the band, so it automatically shifts up to 7075kHz to TX
6. observe SWR goes very high, because the previous tuning 1kHz lower isn't in effect, and probably some previous tuning for another antenna is still in place at that frequency.

I may be triggering this behavior more since I've been switching between 2 different antennas that need very different tuner settings: EFRW vs Tactical Delta Loop.

## Possible Solution
The older versions of your firmware very rarely exhibited this behavior. Maybe they centered ATU settings around the tuned frequency or had wider ranges? I'm not sure. I've not looked at what actually changed.

## Context (Environment)
R1CBU 0.29.2, WSJT-X over USB.

jflinchbaugh, 2025-03-14:
Of course, maybe the behavior's always been the same, and I'm causing the issue myself by retuning between 2 very different antennas. maybe they were close enough before that i didn't notice when i was using an old or new tuning.

jflinchbaugh, 2025-03-14:
Just now, I even had to retune having moved from 7074 to 7074.5. It seems like a tuning should hold for a little wider than that, right? I'll often rely upon being able to find a person calling cq, turn a little to the side, hit the tuner, then come back, so they don't need to hear me tuning over them. This behavior might be making that very difficult.

gdyuldin, 2025-03-14:
Thanks for report, I'll check.

gdyuldin, 2025-03-14:
Could you share your params.db file?

jflinchbaugh, 2025-03-14:
of course. 

[params.db.gz](https://github.com/user-attachments/files/19250290/params.db.gz)

jflinchbaugh, 2025-03-15:
if it turns out that atu table just needs to be cleared, i can try doing that.

BadamK, 2025-03-16:
The Xiegu firmware auto tunes for every 50khz segments,... Does this alternate firm/software tune differently??
 

If it is the same, as Xiegu, then the auto tune would not make a difference on 7074,... ( Auto tuning on 7074 would be saved for  7050 to 7099)

Unless xiegu has band pass filters in place,... Which you can check with a dummy load and turn off the tuner,... A SWR scan will show where the filters are messing things up on the display,... I've found the latest xiegu firmware is not flat across the bands on dummy load,... In you can see where the filters change during SWR scan with TUNER OFF..  It's fairly bad and I'm looking for a new replacement portable rig soon,... 

They didn't fix ALL the menus with RED TEXT either,... 

But I have seen xiegu's tuner require more than one consecutive tune on the same frequency in order to get a lower SWR,...SOMETIMES it takes 2 or 3 auto tunes

gdyuldin, 2025-03-17:
Thanks for the file.  I guess, main reason is that you have had a different ATU values for close frequencies. Selecting ATU logic tries to choose closest saved value using current frequency. If difference exceed 25 kHz - ATU will not be applied. With using single antenna - it's not an issue, if you tune and save values even for every 1kHz, but in your case it might be, that ant 1 was tuned on 7074kHz and ant 2 on 7076kHz. That way if you change VFO to 7075.1kHz - ATU for ant 2 will be used. Anyway - I need some time to check it with a transceiver.

Perhaps, I need to add some logic to saving ATU params (overwrite close values) to prevent or mitigate such cases.

You can use different ATU values for different antennas by changing antenna ID on the app (On changer/RIT/XIT buttons block). Steps:
* Clear atu table on the params.db
* Set Antenna to 1 (top left block atu label will be "ATU1")
* Connect 1'st antenna, tune
* Set Antenna to 2 (text will be "ATU2")
* Connect 2'nd antenna, tune.
* Before or after switching between antennas - don't forget to change antenna within app.

BadamK, 2025-03-17:
If I'm reading correctly,... This alternate software is using 25khz steps to save tuner settings,... Instead of 50khz steps with xiegu firmware,...

I think we need more options for the operator to control:
1.  the bandwidth of steps: 25khz 50khz, 1mhz, 2mhz, full band, Custom ...widths
2.  RESET/CLEAR  current tuned band,... 
3.  RESET/CLEAR ALL tuned bands
4. RESET/CLEAR all tunes between frequency range 
5. MANUAL step thru option after a tune of capacitor/inductor settings for fine tune ability (with OPTION to turn off or bypass tuner bandwidth steps)
6. WEBSERVER console to make tuner all these adjustments,.. (because there's just not enough buttons on the radio)

BadamK, 2025-03-17:
Another option to SWEEP TUNE ENTIRE BAND per operators setting of bandwidth segments

jflinchbaugh, 2025-04-03:
So when in doubt, I activate the tuner where-ever I am. In doing this, I'm pretty sure I'm making a mess of this, even when I'm carefully using one antenna on an "antenna setting" in one afternoon.  I think I'm likely ending up with lots of sets of tuning parameters in pretty arbitrary distances from each other. Now that I know what's happening, I'm constantly thinking, "I need to clear them all and start over," as I get a good SWR in one spot, but bad in another nearby spot.

jflinchbaugh, 2025-04-08:
I watched the database a bit (via serial login) while doing new ATU tunings. I activated ATU on 14073, 14074, and 14075, and the radio clicked lots (changing atu) as I spun the VFO through those frequencies, BUT I noticed the params.db only had the lesser frequency in the DB, and not all 3. I cycled the power on the radio, and now when I spin through those frequencies, it doesn't click so much.  It seems maybe lots of values get into the cache, but not all of them get into the DB. That's good news, I guess, because it means I didn't make a _persistent_ mess of tunings to thrash through from day to day.

jflinchbaugh, 2025-04-09:
based on my previous observation, this may be a fix. it reloads the cache when we update the DB, so we don't end up with every tuning networks cached, even the ones we replaced in the DB. https://github.com/gdyuldin/x6100_gui/pull/144

Maybe this looks viable to you and you can try it out. I need to spend some more time to figure out the build process to test it for myself.

jflinchbaugh, 2025-04-09:
I built and tested my patch, and it appears to work nicely. I can tune in a couple nearby places (7074, 7075, 7076), and I can see the freq move in the atu table of the params.db, as it did before, but when I spin the VFO, it doesn't click in new atu networks in each of those places. That leads me to believe the cache is matching the DB. If I spin more than 25kHz away from the last freq and re-tune, it creates a new entry in the table as expected, leaving the old in place. Then when I spin the VFO, it clicks in new settings between those 2 frequencies as expected as well.

I think the database has been doing mostly the right thing, but the bug was in leaving old values in the cache. #144 fixes that by refreshing the cache from the DB after updating the DB.

commit c10e045c (2025-04-16): Fix saving/loading ATU

Closes #134
closed 2025-04-16 by gdyuldin