OpenX6100/x6100_gui #135: app mode: choose FT8 -> x6100_gui crashes

dl9sau, 2025-03-14:
After upgrading from firmware version 0.25.1 to 0.29.2, I cannot use app FT8 anymore.

Expected Behavior:
fun with FT8

Current Behavior:
Go to App. Press FT8.
"Wait Sync" appears for a few ms, then the firmware version is shown agan.
Reason:
x6100_gui crashes and restarts.

View from Console:
# top
460   266 root     S     194m  19%  20% /usr/sbin/x6100_gui
Go to App -> Press FT8.
488   266 root     S     194m  19%  18% /usr/sbin/x6100_gui
-> new PID. App just dies.

Unfortunately, there's no strace,  ltrace and gdb available at this embeded system. I really recommend it.
So I could only guess that the "last line" in the log shows that something unexpected happens:
```
[root@X6100:/mnt/app_logs]# tail -2 *2025-03* | tail -30
==> x6100_log_2025-03-14_15-52-41.txt <==
[User]	(0.000, +0)	on_vfo_freq_change: Radio set vfo 1 freq=28519500 (28519500) 	(in radio.c line #230)
[User]	(0.000, +0)	on_vfo_momalloc(): corrupted top size

==> x6100_log_2025-03-14_16-04-36.txt <==
[User]	(0.000, +0)	on_vfo_freq_change: Radio set vfo 1 freq=28519500 (28519500) 	(in radio.c line #230)
[User]	(0.000, +0)	on_vfo_momalloc(): corrupted top size

==> x6100_log_2025-03-14_16-04-54.txt <==
[User]	(0.000, +0)	on_vfo_freq_change: Radio set vfo 1 freq=28519500 (28519500) 	(in radio.c line #230)
[User]	(0.000, +0)	on_vfo_momunmap_chunk(): invalid pointer

==> x6100_log_2025-03-14_16-06-16.txt <==
[User]	(0.000, +0)	on_vfo_freq_change: Radio set vfo 1 freq=28519500 (28519500) 	(in radio.c line #230)
[User]	(0.000, +0)	on_vfo_momunmap_chunk(): invalid pointer

==> x6100_log_2025-03-14_16-07-44.txt <==
[User]	(85.050, +5)	lv_finder_create: begin 	(in lv_finder.c line #43)
[User]	(85.140, +90)	on_vfo_freq_change: Radio set vfo 0 freq=14074000 (malloc(): corrupted top size

==> x6100_log_2025-03-14_16-15-18.txt <==
[User]	(443.650, +0)	cfg_band_params_save_all: Save band params for pk=202 	(in band.c line #360)
[User]	(443.650, +0)	cmunmap_chunk(): invalid pointer

==> x6100_log_2025-03-14_16-15-34.txt <==
[User]	(14.430, +0)	cfg_band_params_load_item: Loaded grid_min=-119 (pk=11) 	(in band.c line #433)
[User]	(14.430, +0)	cfg_band_params_load_item: Loaded grid_max=-73 (pk=11) 	(in band.c line #4
==> x6100_log_2025-03-14_16-15-48.txt <==
[User]	(9.300, +330)	cfg_params_save_item: Saved band=11 (pk=0) 	(in params.c line #141)
[User]	(11.635, +2335)	lv_waterfall_create: begin 	(in lv_watmunmap_chunk(): invalid pointer
```

Messages of level "Error" in the log: only these to types:
```
[root@X6100:/mnt/app_logs]# grep Error *2025* |cut -d: -f2-|sort -u
[Error]	(0.000, +0)	cfg_init: Error during loading params 	(in cfg.c line #61)
[Error]	(0.000, +0)	connect: GPSD open: can't connect to host/port pair 	(in gps.c line #30)
```
[No I don't have a GPS connected or configured]

## Possible Solution
Ah. I have an Idea.
[root@X6100:/mnt]# mv params.db params.db.bad; sync ; sync; reboot -fn
;)
System generates new params.db. App -> FT8 works.
But I don't like to loose all my settings. -> I think the current firmware reads something incompatible from the old params.db that leads the app to crash when entering mode FT8.

Steps to Reproduce:
1. you can get my params.db and try

vy 73,
  - Thomas  dl9sau

gdyuldin, 2025-03-17:
You can try to build an app with additional debug flags by uncommenting the last 2 lines in [this](https://github.com/gdyuldin/x6100_gui/blob/main/src/CMakeLists.txt) file, or upload your params.db here.

dl9sau, 2025-03-18:
tnx.
The settings I mentioned I'd have to do again: well, actually they were less than I had in mind.
Anyway, it's good to fix the problem, because
  - we see in the current firmware that an incompatible params.db can crash x6100_gui. -> There should be both, bugfix for the crash conditions (on_vfo_momalloc(): corrupted top size, malloc(): corrupted top size, lv_watmunmap_chunk(): invalid pointer, etc..), and integrity check when parsing the config db.
  - That bug took me only quarter day to fix ;) But think of someone on journey, he might have not been able to use his trx..
 
Here is my defect params.db:
aeh - file type not allowed. hmm, I zip'ed it

[params.zip](https://github.com/user-attachments/files/19323562/params.zip)

gdyuldin, 2025-03-19:
Thanks, I'll check