OpenX6100/x6100_gui #145: Can not build

CGsama, 2025-04-13:
Try to build follow readme but seems missing files.

```
root@a42683c182de:~# cd x6100
root@a42683c182de:~/x6100# cd AetherX6100Buildroot
root@a42683c182de:~/x6100/AetherX6100Buildroot# ./br_build.sh
bash: ./br_build.sh: No such file or directory
root@a42683c182de:~/x6100/AetherX6100Buildroot# ./br_config.sh
make: Entering directory '/root/x6100/AetherX6100Buildroot/buildroot'
make: *** No rule to make target 'X6100_defconfig'.  Stop.
make: Leaving directory '/root/x6100/AetherX6100Buildroot/buildroot'
```

gdyuldin, 2025-04-14:
Hi. I updated README with correct buildroot commands

closed 2025-04-14 by gdyuldin

CGsama, 2025-04-14:
thx for the update!
for the new script it throw error on the last step ./build.sh
```
CMake Error at /usr/share/cmake-3.28/Modules/CMakeDetermineSystem.cmake:176 (message):
  Could not find toolchain file:
  ../../../AetherX6100Buildroot/build/host/share/buildroot/toolchainfile.cmake
Call Stack (most recent call first):
  CMakeLists.txt:3 (project)

CMake Error: CMake was unable to find a build program corresponding to "Unix Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
make: *** No targets specified and no makefile found.  Stop.
```

seems missing 
```
cd build
make
```
after ./br_config.sh

but after the make done, and then do ./build.sh it gives can't find <ft8lib/constants.h>
am i missing something?

gdyuldin, 2025-04-14:
You should install all requirements for building buildroot: https://buildroot.org/downloads/manual/manual.html#requirement

Brandoskey, 2025-04-19:
> thx for the update! for the new script it throw error on the last step ./build.sh
>
> ```
> CMake Error at /usr/share/cmake-3.28/Modules/CMakeDetermineSystem.cmake:176 (message):
>   Could not find [...]

Were you ever able to get this to build?

CGsama, 2025-04-22:
> Were you ever able to get this to build?

yes, need follow that repo to make it first