OpenX6100/x6100_gui #146: FT8 power limit not apply

CGsama, 2025-04-14:
Issue tracker is **ONLY** used for reporting bugs. New features should be discussed on project's [discussions](https://github.com/gdyuldin/x6100_gui/discussions) or on [groups.io](https://groups.io/g/xiegu-x6100/topics)

## Expected Behavior

locked to 5w output

## Current Behavior

after long time operation (~ 12h ), or other reason, the 5w ft8 power limit seems gone

[image]

## Possible Solution

## Steps to Reproduce

1. get in ft8 mode at 20m
2. change to 17m
3. do atu
4. 17m seems out of limit, then other bands

## Context (Environment)

connect to efhw which almost no turn at 20m.
usb connect to pc but no software running  (for grounding)

## Detailed Description

## Possible Implementation

gdyuldin, 2025-04-15:
Hi. I guess, it might be related with ATU. I'll check.
Thanks for your report!

CGsama, 2025-04-15:
Oh, and as you mentioned atu, long press atu button sometime not response and may need tune every time switch band within ft8 mode or will go red, also sometimes no output and spectrum is empty need do a full restart.

gdyuldin, 2025-04-19:
Well, let's create a one issue issue for a one problem. Otherwise it will be hard to track - which already fixed and which - not.
Sometimes buttons doesn't respond - I guess, it's a hardware issue, at least the same sometimes happens with OEM firmware.

gdyuldin, 2025-05-11:
Added command to limit output power before every FT8 TX. Should solve this issue.
Will be in the next release.

closed 2025-05-11 by gdyuldin