OpenX6100/x6100_gui #147: Winlink Express/VaraHF doesn't Switch Frequenzcy on Radio

aendes, 2025-04-18:
When using Winlink Express (v1.7.22) together with VaraHF, I encountered an issue related to frequency switching on the radio.

**Steps to Reproduce:**

Launch Winlink Express.

Open a VaraHF session window — the radio changes frequency as expected.

Select a target station from the dropdown list — the radio does not change to the selected station's frequency.

Close the session window and reopen it — at this point, the radio switches to the correct frequency of the previously selected station.

**Expected Behavior**:
Upon selecting a station from the list, the radio should immediately switch to the corresponding frequency without needing to close and reopen the session window.

**Observed Behavior**:
The frequency only changes on the radio after the session window is reopened. This does not happen immediately after selecting a station.

**Additional Info**:

This behavior was not observed when using the official Xiegu firmware — in that case, frequency changes occur instantly upon station selection.

I attempted various configuration options within Winlink Express but was unable to find a setup that resolves the issue.

Screenshot of my Settings:

[image]

gdyuldin, 2025-04-19:
Hi. Thanks for your report. Could you, please, check "Log radio control..." checkbox, repeat your steps and send me log file (RMS Express ... log). Looks like some commands was not implemented on the CAT.

aendes, 2025-04-20:
The behavior appears to be inconsistent and influenced by whether debugging is enabled. After enabling the debug output, the frequency switching works as expected when selecting a station. However, when disabling debug mode, the issue reappears and the frequency does not switch until the session window is reopened.

I attached the log.

[RMS Express 20250420.txt](https://github.com/user-attachments/files/19824434/RMS.Express.20250420.txt)

gdyuldin, 2025-04-21:
Thanks. I'll take a look. At least it might contains some CAT commands, that I need to take check

closed 2025-06-04 by gdyuldin