OpenX6100/x6100_gui #149: Decode long ft8

CGsama, 2025-04-19:
Issue tracker is **ONLY** used for reporting bugs. New features should be discussed on project's [discussions](https://github.com/gdyuldin/x6100_gui/discussions) or on [groups.io](https://groups.io/g/xiegu-x6100/topics)

## Expected Behavior

decode long message like this
[image]

if URE doing this then it may be valid

## Current Behavior

not show at all

## Possible Solution

## Steps to Reproduce

## Context (Environment)

## Detailed Description

## Possible Implementation

gdyuldin, 2025-04-19:
I have a test for such messages - https://github.com/gdyuldin/x6100_gui/blob/main/tests/test_ft8_qso.cpp#L163 - so it should work. I guess it might be due to signal decoding (current implementation of FT8 lib decodes ~ 80% of messages, comparing wsjtx).

CGsama, 2025-04-22:
i see, i always open the ft8 app and wsjtx(decode only) together, but never see long sentence, while pretty much all other message from same station are picked at ~0db, maybe the lib issue only do 77 bits

gdyuldin, 2025-04-24:
77 bit is a protocol limitation. In theory, resampling signal with more bins might improve receiving of some signals. There is a python implementation on the ft8_lib, that do it and it shows better recall. Unfortunately, this logic is not fully implemented on the C.

closed 2025-05-11 by gdyuldin