OpenX6100/x6100_gui #150: Non-standart call-signs (like /QRP) ft8 app no tx

gdyuldin, 2025-04-19:
### Discussed in https://github.com/gdyuldin/x6100_gui/discussions/148

<div type='discussions-op-text'>

<sup>Originally posted by **aidiesha** April 19, 2025</sup>
Hi
I just tried to put the / after my call example 9w2di/qrp. But it does not tx and it's keep floating 

Regards</div>

OE3BCB, 2025-04-23:
Hello i have the Same Problem.
When i put ..../QRP it does not send. I have more than 10 CQ ..../QRP Messages in the FT8 Window.

73 de Chris

gdyuldin, 2025-04-23:
No, CQ modifiers and non-standard callsigns encoded in different ways.

gdyuldin, 2025-05-21:
There is a bug in the ft8lib. Seems, I fond a way to fix it, need add some tests.

closed 2025-05-21 by gdyuldin

dk5cr, 2025-05-21:
Thanks for fixing. Can I replace the ft8 lib to test it? Or do you have an image? 
73

dk5cr, 2025-05-21:
Sorry, webbrowser showed the site from the cache, now got it. 
It works! Thanks

commit da9c45bd (2025-05-21): Add support of the non-standard callsign

Closes #150