OpenX6100/x6100_gui #153: Reporting 0% on Full battery as well as limited to 5w on battery power

jackwangxyw, 2025-05-01:
## Expected Behavior
Using v0.29.2 btw. It should on the top right of the screen where it displays the time also display the voltage and percentage of the battery correctly. Also it should be transmitting at full 10w power on the in built battery.

## Current Behavior
Instead of a correct percentage of the battery, it just shows 0%. (The voltage seems to be correct) Also it seems to be limited to 5w of output on the internal battery.

## Steps to Reproduce

[image]
[image]

1. The radio is pretty hot, was doing a ssb net with the local club on 10w. The back was definitely uncomfortable to the touch. 
2. The radio also seemed very slow when trying to access the webserver, taking a long time to load and sometimes outright failing.

## Context (Environment)
One of the main reasons for choosing this firmware was being able to do 10w on internal battery, kinda disappointing that it doesn't work.

gdyuldin, 2025-05-01:
Battery percentage is reported by MCU on the BASE (radio) board within transceiver. You will see the same using OEM firmware. It can't be fixed.
Regarding 10W while using internal battery - it's not a software limitation. Even 8.4 volt (maximum voltage of internal battery) is not enough for output transistors to achieve 10W.

closed 2025-05-01 by gdyuldin

jackwangxyw, 2025-05-01:
can you override the reported percentage and calculate based on voltage? Since it seems to be reporting the right voltage, it should be simple to calculate the percentage of the battery based on that. I'm not a software dev, so idk how hard that would be to implement