OpenX6100/x6100_gui #155: can't save the new qso data into the "ft_log.adi" file. Please help me!

Yoshi2025jp, 2025-05-10:
I am using 0.29.2 of the x6100. Until the other day, After the FT8 QSO, I downloaded ft_log.adi to my PC via wifi and uploaded it to eqsl and qrz.com, but it suddenly stopped working. I downloaded 0.29.2 again today, saved it to the SD card with rufas, started it up, and now I can't save the ft8 communication log. The logs from the other day are still there, but the QSOs from today are not being recorded. I did "factory reset" the x6100, reformatted the SD card, and created it with rufas, but it still doesn't work. The QSO data from today is no longer saved to ft_log.adi. I don't know what's wrong. Someone please help me. Thank you. (The text may be a little strange because it was translated into English with Google Translate.)

gdyuldin, 2025-05-10:
Hi. Writing image to SD card is not completely wipe it - settings are preserves.
You can insert SD card to the PC and delete ft_log.adi on the DATA partition. Sometimes Windows assign a letter only for the first (BOOT) partition. If you see only one partition in the explorer with boos.scr and 2 another files - open partition manager and manually assign a letter to the DATA partition (3rd on the SD card)

Yoshi2025jp, 2025-05-10:
Thank you for your quick response. I assigned a drive letter as you suggested, and was able to find ft_log.adi. When I read it directly on my PC, I was able to confirm that the data was saved safely, but when I download it from a browser using the x6100's web server, new communication records are not saved, which is troubling me. Would it be better to release all three allocations on the SD card and then use rufas to create a new SD card? The web server function is very useful, so I would like to continue using it in the future. Thank you.

Yoshi2025jp, 2025-05-11:
As I wrote earlier, I deleted the three partitions on the SD card and then created an SD card using rufas. Then I inserted it into the x6100 and started the radio. Unfortunately, the ft_log.adi downloaded from the web server still had update data from a few hours ago, and the data I updated with several stations after that was gone. It seems that even if I completely wiped the contents of the SD card, the x6100 itself or my PC connected via wifi read the past update data, and I just can't figure out what's going on. When I looked at the data folder on the SD card directly on my PC, I found that files such as ft-log.adi that I had formatted and deleted were still there with the previous timestamps. Please help. Thank you.

Yoshi2025jp, 2025-05-11:
I'm replying to myself. I'm sorry for worrying you, but the problem seems to have been resolved. I frequently display past data, so I tried deleting the cache on the MS Edge browser, and new data was successfully downloaded from the x6100 web server!! I hadn't paid much attention to such basic things, so it took me a while to resolve the issue. I'm very sorry that the updated data for about 10 stations has disappeared, and I'm sorry to the operators of those stations. I can't upload data to eqsl or qrz.com. I appreciate everyone's concern. Thank you.