OpenX6100/x6100_gui #163: Move PWR to MFK please,...

KbCr2iKaJnQ, 2025-06-15:
I use VOL most ALL the time on Vol knob,...

I use PWR almost the same amount and it needs to be moved to MFK encoder,...

I don't want to be switching knob function every time I use my MOST USED functions.

Hope that makes sense,... 

Thanks for all the time and support for this project 

It is GREATLY appreciated.

gdyuldin, 2025-06-15:
Hi.
Application architecture makes such change very hard. All controls are "tied" to the VOL and MFK encoders, there are a lot of time should be spend to refactoring to do it.

gdyuldin, 2025-06-27:
When any application is opened - VOL encoder controls only volume. When you change volume when your FT8 app is opened - you hear change without any distracting visual indication. Allowing more functions requires to adding/enabling visual notifications for current control and it's value, which will hides some application's information. 

@neolinux, according your proposal both VOL and MFK will be busy (for volume and output power control). But for FT8 MFK encoder used to navigate through the table. Application can't use same encoder for both functions.

gdyuldin, 2025-06-27:
I guess, that ability to assign any 2 parameters at the same time for both encoders is the root cause of this issue. Just moving PWR to MFK (of hardcode Volume, SQL and RFGain to VOL) is not a best solution. Someone might agree with your proposal, others - not.

As I mentioned above - there are many places, which should be rewritten to make assignings more flexible. At a glance this task requires:

* Create a single enum for all parameters (VOL + MFK)
* Make a new function (or class instance) for each parameters
* Change calling logic of each parameters, perhaps using map of functions/instances
* Change logic of saving which parameters are added to fast access with encoders
* Change a logic of adding parameter to fast access
* Create a migration to update already saved parameters to a new logic

closed 2025-07-05 by KbCr2iKaJnQ