OpenX6100/x6100_gui #169: Knobs info works GREAT,... (DNF:On/Off  does not display)

KbCr2iKaJnQ, 2025-06-29:
Love the function of "Knobs info",...

If you notice any selection for MFK seems to have "Double Entries",... 
Remove DNF:(On/Off) function from MFK encoder,...

DNF does not display with a LABEL,... which causes one of the other displayed options for MFK to display twice,...

Give DNF a Label for Encoder entry,... or remove ability of adding DNF:on/off to encoder.

Still need to have TX POWER enabled for MFK,...(will keep making mention of this until someone can make it happen,... (TX Power is on the MFK on Xiegu Firmware,... it can be made to function on MFK for alternative also)

Thats all for now,...
Great job to everyone that supports this project,... 

If I could find time to learn the programming code, I would attempt to input other options.
Thank you all,

Issue tracker is **ONLY** used for reporting bugs. New features should be discussed on project's [discussions](https://github.com/gdyuldin/x6100_gui/discussions) or on [groups.io](https://groups.io/g/xiegu-x6100/topics)

## Expected Behavior

## Current Behavior

## Possible Solution

## Steps to Reproduce

1.
2.
3.
4.

## Context (Environment)

## Detailed Description

## Possible Implementation

gdyuldin, 2025-06-30:
Thanks, I'll remove long press actions for ON/OFF controls

Please, don't mix many things on a single issue. 
Also, please, don't create a duplicate of your issue - #163

KbCr2iKaJnQ, 2025-06-30:
This seemed only to be affected for DNF on/off,...

And I did not add it to the encoder,... It was already registered with MFK from a fresh install of 0.31.0,...

I've removed it each time I did a reinstall of 0.31.0
And it still is registered to MFK on rewriting sd

closed 2025-07-01 by gdyuldin

KbCr2iKaJnQ, 2025-07-02:
DFN Works ok now in 0.31.1
(and, no control Vol has corrected itself also in 0.31.1)

commit 1bc746f4 (2025-07-01): Add on/off controls to knob info

Closes #169