OpenX6100/x6100_gui #172: KNOB info issue,... No VOL control after reboot?

KbCr2iKaJnQ, 2025-06-30:
I've done a fresh reinstall,... (Removing all data part files before rewriting SD IMAGE) 

Booted up just fine,... (Even after using ETCHER for Android device)

I still had to remove DNF from MFK knob for display issue,... (Noted on another issue)

Also adjusted some of my other settings that got defaults from data dump,...

After rebooting, I can not adjust VOL or cycle to it with knob,... Can only access VOL from menu,.

And,, can not add or remove VOL to encoder.
And no change by disabling KNOB INFO setting.

Other VOL encoder items seem to work and cycle OK.

KbCr2iKaJnQ, 2025-06-30:
VOL worked correctly  after fresh install,... But not after rebooting to save settings,...

gdyuldin, 2025-06-30:
Please, use a template to create a new issue.
Issue should contain only single problem, not a list of them.
Issue should has a short but complete description (steps to reproduce).

I rebooted my device and VOL is working as expected.

KbCr2iKaJnQ, 2025-06-30:
This was one issue with VOL not registering with VOL encoder after a reboot,... Just described what I did,...

Which wasn't much,... It was a fresh install...