OpenX6100/x6100_gui #177: Can split the config option of I/Q offsets to each band?

d3m3vilurr, 2025-07-27:
I followed your [guide](https://www.youtube.com/watch?v=8rzYoUmX2-o).
In my test, it looks to require different setting per band.
that this reason, I reverted offsets to 0.

idk it's normal case or not. if it's normal, gui should support setting per band (or range of the bands..)

gdyuldin, 2025-07-28:
If I understand correctly, there is a logic within OEM BASE firmware, that reads iq offsets for different bands from some memory area during work. Perhaps, all transceivers is calibrated on the factory, or changing band rewrites iq offset. I'll check.

d3m3vilurr, 2025-07-28:
my TX I/Q values

| band | I offset | Q offset |
| ----- | -------- | --------- |
| 160m | -10 | -11 |
| 80m | -8 | - 10 |
| 40m | -4 | -8 |
| 30m | -1 | -7 |
| 20m | 4 | -6 |
| 17m | 8 | -2 |
| 15m | 12 | -7 |
| 12m | 17 | 2 |
| 10m | 26 | 6 |
| 6m | 78 | 0 |