OpenX6100/x6100_gui #186: 10W override for FT8 app

sysadminpower2019, 2025-09-15:
I know there are probably a good number of reasons why it is currently limited to 5w but none the less if you connect a phone / laptop to the radio you can make it do 10W FT8 anyway even without an external power supply. 

It would be nice to have an override option to set to 10W if someone really wanted to

gdyuldin, 2025-09-19:
This device was not designed to work with full output power with modulation like FT8 - I know many cases, when transceiver was damaged after this. If you want more power - you can use external amplifier.

sysadminpower2019, 2025-09-19:
Well i personally have had no issues with 10W on digital modes but i
understand the concerns.

Well the x6200 goes up to 8W and is similar so maybe we could allow it up
to 8W